sipeed/NanoKVM-USB #3: MacBook Pro - Only device listed is MacBook camera

danielmbrown603, 2025-03-09:
Host = MackBook Pro M4
Target = Mini PC

I tried safari and chrome browsers. The MacBook only shows the MacBook camera as an available device.

wj-xiao, 2025-03-10:
Please try the following troubleshooting steps:

1. Ensure all cables are correctly connected: [doc](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM_USB/quick_start.html#Wiring)
2. Confirm that your Mac has granted the permissions to the device: [doc](https://support.apple.com/guide/mac-help/use-the-ports-on-your-mac-mchlf779ae93/mac)
3. As a test, attempt to control your Mac from PC
4. Use a different, known-good USB cable to eliminate the possibility of a cable fault

If you continue to experience problems, please contact us at `support@sipeed.com`.

danielmbrown603, 2025-03-15:
> Please try the following troubleshooting steps:
>
> 1. Ensure all cables are correctly connected: [doc](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM_USB/quick_start.html#Wiring)
> 2. Confirm that your [...]

Thanks for the help, it was the permission issue noted in step #2. After that was resolved, it worked and works well, 

Thank you

closed 2025-03-15 by danielmbrown603