sipeed/NanoKVM-USB #5: Just release the src code

rtfmkiesel, 2025-03-11:
Stop toying around with open-source development. In the interest of security and usability, release the code **now** and not when the repo has N stars. This behavior is embarrassing for FOSS development.

My points:
+ Security: Given your track record on the regular NanoKVM, document **ALL** features of your devices. For example, I would like to have documentation on the events/commands
    + `READ_MY_HID_DATA`
    + `GET_PARA_CFG`
    + `SET_PARA_CFG`
    + `GET_USB_STRING`
    + `SET_USB_STRING`
+ Usability: I have a non-english keyboard and need to be able to adjust the keyboard commands, as special keys currently do not work for me. Reversing your `app.asar` should not be necessary.

Zepan, 2025-03-14:
Hi, the NanoKVM-USB is an offline USB peripheral device, which inherently poses no security risks. 
This repository is primarily used for collecting user feedback and bug reports, as evidenced by the issues raised by other users.

We are more than willing to release the code earlier,  if users approach us with genuine technical inquiries and constructive discussions. 
However, if individuals fueled by bias and conspiracy theories demand that we open-source the code under the guise of "security concerns," we will not entertain such requests. 
Their goal is not truly about "security" or "open-source" principles, but rather the thrill of online bully.

When it comes to networked IP KVMs, they may have some grounds to raise questions, but to extend such skepticism to an offline USB KVM is unreasonable. What’s next? Endless baseless accusations? We will not be swayed by such tactics.

WarmWelcome, 2025-03-14:
> Hi, the NanoKVM-USB is an offline USB peripheral device, which inherently poses no security risks.

Offline USB devices can absolutely pose security risks. Here are just a few that easily demonstrate that fact:
https://docs.hak5.org/hak5-usb-rubber-ducky
https://usbkill.com/
https://www.hackmod.de/en/p/inputstick-usb

> This repository is primarily used for collecting user feedback and bug reports, as evidenced by the issues raised by other users.

Please modify the readme to state this as being the case. How is everyone is supposed to assume that this is for issues and feedback only by the fact that a select few people have posted issues?

> We are more than willing to release the code earlier, if users approach us with genuine technical inquiries and constructive discussions.
> However, if individuals fueled by bias and conspiracy [...]

...So what you're saying is that you're going to delay the release of the code because people are asking for the release of that code? That makes no sense, sorry. Could you explain in better terms what you mean to do here by delaying the release of the code that would prove that this project is indeed secure?

Our requests for posting the source code for security reasons isn't based on "bias and conspiracy theories" but on legitimate concerns, given that you and another collaborator have acknowledged several potential security issues and flaws ([your statement](https://github.com/sipeed/NanoKVM/issues/301)), as well as outright ignoring security issues ([collaborator statement](https://github.com/sipeed/NanoKVM/issues/301#issuecomment-2641756220)).

If you won't share the code openly, at least have the decency not to dismiss our legitimate concerns with accusations of bullying and conspiracy. Stating that those who ask for the source code are conspiracy theorists is not only misguided, but also incredibly dismissive of valid concerns. And to state that these requests for open source are bullying throws out constructive conversation and trust. Shutting people down with this language and these labels is not appropriate. When legitimate concerns about security are dismissed as mere conspiracy theories or bullying, trust evaporates. Trust isn't built by name calling and labeling people with concerns as bullies, but by addressing those concerns.

rtfmkiesel, 2025-03-14:
I'm not one for pointless discussion about conspiracy theories. I would just like the source code to a device I own, if the manufacturer intends to share the code anyway. This (usually) includes documentation about its features. 

+ As I said in the original comment, I need to adjust the keymap. This is the main reason I want to take a look at the source code for the serial connection. (Both on the microcontroller and the web interface)
+ Also, I already added a USB jiggler button to my web interface, which is a neat feature I think others should benefit from too. 

Personally, @WarmWelcome, I think it is a bit far-fetched calling this a rubber-ducky like device without any hard evidence. In that case, you would have to call every keyboard manufactured in *insert bad country here* a backdoored device, too. The only way to trigger this would be by either a random interval, or via a specific serial command issued through USB. Since the [web interface part](https://cdn.sipeed.com/nanokvm/NanoKVM-USB.zip) is completely usable offline and actually does not make any outside connection. (see the image below) Thus it's hard to believe the manufacturer has a way of remotely talking to those devices.

[image]

patschi, 2025-03-31:
> We are more than willing to release the code earlier, if users approach us with genuine technical inquiries and constructive discussions.

My two cents, @Zepan...

I pre-ordered NanoKVM USB because I do really appreciate your effort and work you're putting into the entire KVM-lineup. Additionally I also like the idea and it could be very useful in the future.

I also usually defended NanoKVM on a objective-basis in certain discussions - but the "After 1000 stars we open-source it" throws me and - a lot of people I know - off.  I had some minor understanding the first time with the main-NanoKVM... But honestly, this time, this move is rather disappointing.

You have opensourced the main NanoKVM source code already, so I assume the majority of the code basis is already open. There have been great improvements and contributions from random people in your software - why not for this NanoKVM version too? This has so much potential. I'm sure there're people interesting into building more for this too.

You got quite some attention with your lineup. Why don't open-source it straight away and give people an additional reason to buy it and improve it?

patschi, 2025-04-02:
I'm genuinely positively surprised and thankful for Sipeed deciding otherwise - the source code for the browser-part was published ~10 hours ago: https://github.com/sipeed/NanoKVM-USB/tree/main/browser

Even locking the source code behind a specific amount of stars was removed: https://github.com/sipeed/NanoKVM-USB/commit/66cc306c409144417eb1ba314c6239f636a264c5

I'd like to thank you, @Zepan and @wj-xiao, for deciding otherwise ❤

wj-xiao, 2025-04-03:
All codes are now available.

patschi, 2025-04-03:
> All codes are now available.

@wj-xiao Where is the source code for the backend? Is it simply the "normal" NanoKVM image, with solely the frontend being different?

rtfmkiesel, 2025-04-07:
You probably mean the code for the hardware. There is not really a "backend" as the "frontend" (web) controls the device by sending commands over serial (via WebSerial). The screen is displayed via a simple HDMI capture card. 

What's the plan about the firmware for the hardware itself? @wj-xiao

patschi, 2025-04-07:
> You probably mean the code for the hardware.

I mean the server/backend (as they [called it with NanoKVM](https://github.com/sipeed/NanoKVM/tree/main/server) as well). I'm sure there is some backend code responsible for grabbing the HDMI output and processing any commands from serial and send it over HID.

rtfmkiesel, 2025-04-07:
Not really. Chrome is able to display the video signal and directly communicate with a serial device without any additional software. The desktop app is using Electron which simplified is a packed & portable Chrome browser.

wj-xiao, 2025-04-08:
> What's the plan about the firmware for the hardware itself?

There is no firmware code. For more information, please refer to [CH9329](https://www.wch-ic.com/search?t=all&k=9329).

closed 2025-04-08 by rtfmkiesel

WarmWelcome, 2025-04-14:
Thank you for clearing this up.