sipeed/NanoKVM-USB #10: Keyboard not detected by BIOS/EFI during boot sequence

DigitallyRefined, 2025-03-27:
Keyboard and mouse works correctly when booted up into Windows or Debian, however when I reboot into the BIOS the keyboard is not detected, but the mouse is.

It also appears to affect the Linux GRUB menu, but works fine when an OS has started.

I think it's the same issue that affected the NanoKVM in https://github.com/sipeed/NanoKVM/issues/410 is it possible to apply the same fix for the NanoKVM-USB?

Tested on a HP EliteDesk:

https://github.com/user-attachments/assets/1afb5b65-64d8-4a68-8f7d-d6725b3e0f97

https://github.com/user-attachments/assets/31d5dc2d-1413-41f7-88e1-dc0b4ed7d456

https://github.com/user-attachments/assets/cebba6b5-b18c-499e-9033-056184fc0fdb

Z2Z-GuGu, 2025-04-01:
Hello, I actually found this issue on my own HP Elite mini PC as well. However, since NanoKVM-USB uses a different USB-HID scheme, we can't fix this problem for now. We will be arranging some tests for various HID modes soon in an attempt to resolve it. Regardless of the outcome, we will notify you here after the tests. Thank you!

chrislockwood, 2025-05-28:
Any progress in this issue? 

This is my primary reason for not buying.

As time passes I am starting to wonder if the problem might be insurmountable on the current hardware revision, hopefully not the case.