sipeed/NanoKVM-USB #11: Host webinterface on device

patschi, 2025-03-31:
It would be awesome to host the entire web-UI on the device itself, allowing users to access the device's webinterface without internet connection on the online-version nor installing the standalone version.

This might be possible to realize by emulating a network interface towards the host using DHCP, allowing the user to use a specific hard-coded IP to access the webinterface. (via RNDIS?)

geosp, 2025-04-10:
You can host the NanoKVM device within your own network by connecting it to an existing host system, allowing for comprehensive remote control of all your devices through this solution: https://github.com/sipeed/NanoKVM-USB/issues/16#issuecomment-2787460728
This implementation offers significant advantages:

1. Configure your own Tailscale account for secure remote access.
2. Enable xRDP on the host where the device is attached
3. Gain full KVM access to any device connected to your multi-host KVM switch

I've extended this solution by developing a custom remote web application that provides intuitive interface controls for switching between hosts remotely, eliminating the need for physical access to the KVM switch. This creates a truly location-independent remote management solution for your entire hardware infrastructure.

wj-xiao, 2026-01-21:
NanoKVM-USB does not involve network operations and cannot run services. If you require this functionality, please refer to [NanoKVM](https://github.com/sipeed/NanoKVM).

closed 2026-01-21 by wj-xiao