sipeed/NanoKVM-USB #16: Add docker version

maxx-ukoo, 2025-04-04:
It is good to have release with docker image instead run browser locally. 
Working Dockerfile:

```
FROM node:23-alpine3.20

RUN apk add --no-cache wget git
RUN wget -qO /bin/pnpm "https://github.com/pnpm/pnpm/releases/latest/download/pnpm-linuxstatic-x64" && chmod +x /bin/pnpm

RUN git clone https://github.com/sipeed/NanoKVM-USB.git /app && \
    cd /app/browser && \
    pnpm install && \
    pnpm build && \
    npm install -g http-server

FROM nginx:1.27.4-alpine    

COPY --from=0 /app/browser/dist /usr/share/nginx/html
```

maxx-ukoo, 2025-04-04:
Docker image created and pushed - https://github.com/[maxx-ukoo/usbkvm-docker](https://github.com/maxx-ukoo/usbkvm-docker)

You are free to add this to this repo

geosp, 2025-04-08:
Here is a docker version:

```yml
services:
  web:
    image: nginx:alpine
    ports:
      - "8585:443"
    volumes:
      - ./app:/usr/share/nginx/html
      - ./nginx.conf:/etc/nginx/conf.d/default.conf
      - ./ssl/cert.pem:/etc/nginx/ssl/cert.pem
      - ./ssl/key.pem:/etc/nginx/ssl/key.pem
    restart: unless-stopped
```

Generate SSL certs generate_certs:

```bash
#!/bin/bash

# Create directory for SSL certificates
mkdir -p ./ssl

# Generate a self-signed certificate for jetson04
openssl req -x509 -nodes -days 365 -newkey rsa:2048 \
  -keyout ./ssl/key.pem \
  -out ./ssl/cert.pem \
  -subj "/CN=jetson04" \
  -addext "subjectAltName=DNS:jetson04,IP:127.0.0.1"

echo "Self-signed SSL certificates created."
echo "cert: ./ssl/cert.pem"
echo "key: ./ssl/key.pem"
```
nginx.conf
```yml
cat nginx.conf 
server {
    listen 80;
    server_name jetson04;
    
    # Redirect HTTP to HTTPS
    return 301 https://$host:8585$request_uri;
}

server {
    listen 443 ssl;
    server_name jetson04;
    
    # SSL configuration
    ssl_certificate /etc/nginx/ssl/cert.pem;
    ssl_certificate_key /etc/nginx/ssl/key.pem;
    
    # Improved SSL settings
    ssl_protocols TLSv1.2 TLSv1.3;
    ssl_ciphers HIGH:!aNULL:!MD5;
    ssl_prefer_server_ciphers on;
    ssl_session_cache shared:SSL:10m;
    ssl_session_timeout 10m;
    
    # Root directory and index file
    root /usr/share/nginx/html;
    index index.html;
    
    location / {
        try_files $uri $uri/ /index.html;
        
        # Enable CORS for all origins
        add_header 'Access-Control-Allow-Origin' '*';
[... 558 lines cut ...]
    <div id="remote-button" onclick="openKvmRemote()">
      <span>Remote</span>
    </div>
    
    <script>
      function openKvmRemote() {
        // Set popup dimensions
        const width = 500;
        const height =900;
        
        // Calculate position to center the popup on screen
        const left = (window.innerWidth - width) / 2;
        const top = (window.innerHeight - height) / 2;
        
        // Open the popup window
        window.open(
          'remote.html',
          'KVM Remote Control',
          `width=${width},height=${height},left=${left},top=${top},resizable=yes,scrollbars=yes,status=no,location=no,menubar=no,toolbar=no`
        );
      }
    </script>
  </body>
</html>
```

geosp, 2025-04-08:
This is the final product NanoKVM-USB with KVM Display Remote:

[image]

geosp, 2025-04-10:
This the KVM Switch remote control:

```html
<!DOCTYPE html>
<html lang="en">
<head>
  <meta charset="UTF-8">
  <meta name="viewport" content="width=device-width, initial-scale=1.0">
  <title>KVM Remote Control (Dark Theme)</title>
  <link href="https://cdn.jsdelivr.net/npm/tailwindcss@2.2.19/dist/tailwind.min.css" rel="stylesheet">
  <style>
    /* Dark-themed button styles using custom CSS (Tailwind @apply directives replaced with CSS rules) */
    .btn-primary {
      background-color: #1D4ED8; /* dark blue */
      color: #ffffff;
      font-weight: bold;
      padding: 0.5rem 1rem;
      border-radius: 0.375rem;
      transition: background-color 0.3s;
    }
    .btn-primary:hover {
      background-color: #1E40AF;
    }
    
    .btn-secondary {
      background-color: #374151; /* dark slate gray */
      color: #D1D5DB; /* light gray */
      font-weight: bold;
      padding: 0.5rem 1rem;
      border-radius: 0.375rem;
      transition: background-color 0.3s;
    }
    .btn-secondary:hover {
      background-color: #4B5563;
    }
    
    .btn-success {
      background-color: #16A34A; /* green-600 */
      color: #ffffff;
      font-weight: bold;
      padding: 0.5rem 1rem;
      border-radius: 0.375rem;
      transition: background-color 0.3s;
    }
    .btn-success:hover {
      background-color: #15803D; /* green-700 */
    }
    
    .btn-danger {
      background-color: #DC2626; /* red-600 */
      color: #ffffff;
      font-weight: bold;
      padding: 0.5rem 1rem;
      border-radius: 0.375rem;
      transition: background-color 0.3s;
    }
    .btn-danger:hover {
      background-color: #B91C1C; /* red-700 */
    }
    
    .btn-active {
      background-color: #1E3A8A; /* blue-800 */
      color: #ffffff;
      font-weight: bold;
      padding: 0.5rem 1rem;
      border-radius: 0.375rem;
      box-shadow: inset 0 2px 4px rgba(0, 0, 0, 0.2);
    }
    
    .btn-disabled {
[... 352 lines cut ...]
    });

    applySettingsBtn.addEventListener('click', applySettings);

    computerBtns.forEach(btn => {
      btn.addEventListener('click', () => {
        const pcNumber = parseInt(btn.dataset.pc);
        switchToComputer(pcNumber);
      });
    });

    autoSwitchToggle.addEventListener('change', toggleAutoSwitch);

    autoSwitchIntervalSlider.addEventListener('input', () => {
      autoSwitchInterval = parseInt(autoSwitchIntervalSlider.value);
      intervalDisplay.textContent = `${autoSwitchInterval}s`;
      timeRemaining = autoSwitchInterval;
      updateCountdown();
    });

    disableComputerButtons();
  </script>
</body>
</html>
```

julio-matarranz, 2025-09-04:
Dear,
Great idea, I was looking for Docker image.
Could you make a pull request on the project ? 🤔

closed 2025-10-30 by wj-xiao