sipeed/NanoKVM-USB #20: Windows 11 ARM support

cowmix, 2025-04-13:
Please update your documentation to include that users of Windows 11 ARM must install this USB serial driver before using the web app / page. The serial USB driver is missing in the ARM build of Windows.

https://www.wch.cn/downloads/CH341SER_EXE.html

alryaz, 2025-05-15:
AFAIK it is possible to extract the CH341 driver executable and manually install the .inf file.

wiiplay, 2025-10-30:
Hi @cowmix, I tried the driver from (direct link) [](https://www.wch.cn/downloads/ch341SER_EXE.html?type=en) and Windows immediately picked up the serial port on the NanoKVM USB

wj-xiao, 2026-01-21:
Thanks for your feedback. We have added instructions in the [document](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM_USB/faq.html#No-USB-Serial-%28COMx%29-Device-After-Opening-Web-Page-on-Windows).

closed 2026-01-21 by wj-xiao