sipeed/NanoKVM-USB #38: NanoKVM-USB target machine KB/Mouse not working

illusive5943, 2025-05-06:
Hello, 

I have tried Chrome browser https://usbkvm.sipeed.com/ and the desktop client 1.0.1. 

COM port is properly recognized and I have display, but no KB/M input. Also tried 2 different USB-ports on target machine. It's a Lenovo laptop.
The built-in onscreen keyboard does not work either

Any help?

[image]

charlievogt, 2025-05-12:
I have the same issue.

itzikbenabu, 2025-06-11:
Hi,
I have the same problem on Windows 11. Additionally, I see an error in the Device Manager for USB Serial, which is related to the NanoKVM USB.

[image]

itzikbenabu, 2025-06-12:
The problem is that in Windows 11, it looks like CH341 is not HCL regarding the UEFI bios, the firmware contains support.